nightscout/AndroidAPS #3904: Manually added BG checks < 39 will be amultipled by 18

cr582, 2025-03-25:
Assuption is that you set up your AAPS 3.3.1.3 with mg/dl.

Go to actions tab, click BG CHECK, enter any number between 39 and 36.  BG is multiplied by 18. EG if you enter 36 it will be displayed as 648mg/dl - so obviously there's something wrong.

Please also see [facebook thread](https://www.facebook.com/groups/AndroidAPSUsers/posts/4076035892617797/?comment_id=4076433395911380) by Adrienne Denges, She added a short mp4 movie and screenshots.

Thanks in advance
Chris

cr582, 2025-03-27:
it is also reproducible in 3.3.2.0

olorinmaia, 2025-03-29:
Confirming. It's logged correctly in graph in AAPS, AAPSClient and NS, also in NS treatments. So that is good. So it's a minor bug related to what we see in dialogue box.

olorinmaia, 2025-04-01:
Fixed in latest dev. This issue can be closed
https://github.com/nightscout/AndroidAPS/pull/3918#event-17077654954