nightscout/AndroidAPS #3917: Likely bug in sending duration for profileSwitch, mongodb has different duration value

shoaib42, 2025-03-29:
While extracting data from mongodb, and going through {eventType : Profile Switch}. I notice that the duration is not properly represented in millseconds and seems like there is a dual conversion from minutes to millseconds `(60*1000 * 60*1000)`. As can be seen in the screenshots below, the duration on mongodb is reported as `2592000000000` which equals to `720 mins` if we divide `2592000000000 / (60*1000 * 60*1000)`. I'm unable to place it in the code, tried my best to find the root cause of this. I wonder if this bug is related to profiles not switching off.

The version of AAPS is 3.3.1.3

[image]

[image]

vanelsberg, 2025-04-01:
Hmmm... just curious: at what timezone are you running?

shoaib42, 2025-04-02:
I'm running on America/Toronto (EST/EDT). I have noticed that Profiles either get America/Cuiaba or America/Panama based on the day light savings, although the utc offset is properly set in these timestamps as either -0400 or -0500. For all the profile switch documents I see that the duration has this "Dual conversion" of minutes to milliseconds.

MilosKozak, 2025-04-05:
fixed by https://github.com/nightscout/AndroidAPS/commit/9b90eea7740e86d8e614ce3b6b98a912348e2668
Thanks!

closed 2025-04-05 by MilosKozak