nightscout/AndroidAPS #3920: Medtrum Nano - on connection loss while bolusing, carbs are not recorded, but insulin is in full amount

miiihi, 2025-03-29:
Using Medtrum Nano, giving bolus from the calculator, every once in a while I get a connection loss (probably around every 10th bolus). The connection loss is not a big problem itself, although might be good to look into it too...

The problem is, that the carbs from the calculator are only recorded at the end of the successfull bolus, meaning they are not recorded on connection loss... although the insulin is recorded in full amount !?!?

IMO, the carbs should be recorded immidatelly after confirmation, same as insulin... 

Thanks, Blaž

MilosKozak, 2025-03-31:
Yeah, carbs are recorded only on successful bolus. It's by design and it's safety feature

miiihi, 2025-03-31:
Hmm, I don't understand how is carb recording related to successfull / unsuccessfull bolus and how is this safety feature... IMO this is a bug. I have eaten the same amount of carbs unrelated to the bolus being successful or unsuccessful.

If this indeed is going to stay as "safety" feature, then the notification text should be adjusted. The text is saying sth like 'Bolus error... please check the amount of insulin delivered..." The notification should mention, that the carbs were not recorded!

MilosKozak, 2025-04-05:
Easy. If bolus fail and carbs are recorded and user enter the carbs again in wizard it will get bolus acording to double amount of carbs

MilosKozak, 2025-04-05:
i see there this message "Bolus reported an error. Manually check real delivered insulin and carb amount"

closed 2025-04-05 by MilosKozak