nightscout/AndroidAPS #3931: DASH no longer shows the "silence alert" button to disable pod is alerting beeps

vanelsberg, 2025-04-03:
**AndroidAPS 3.3.3.0-dev**
**Build**: 4d18c1a2
**Flavor**: Fullrelease

When the DASH pod is alerting (_expiration reminder, shutdown, low reservoir_) a button should appear on the DASH tab för user to "silence" the Pod beeps. This button is no longer enabled (made visible). See WiKi [Silencing Pod Alerts](https://wiki.aaps.app/en/latest/CompatiblePumps/OmnipodDASH.html#silencing-pod-alerts)

**Problem description:**
The "Silence Alerts" button on DASH tab is not visible on when Pod is alerting (even after refreshing state).
User can not silence Pod beeping every X minutes.

**Additional information**
Noticed this problem also on [build commit 40068c81ffd5bff07c98763f84cee95f5cb78acc](https://github.com/nightscout/AndroidAPS/commits/dev/) but the problem may also be on earlier commits.

vanelsberg, 2025-04-06:
Update: after reopening AAPS and moving to the DASH tab, very briefly (less then 0.1 secs?) seeing a "flash" of the "silence" button.
(I'd say the button view gets enabled but then on screen refresh when opening the DASH tab the button view is disabled again?)

@MilosKozak Could this be cause by one of the dev library updates done recently? Not sure where to look.... 🤔

Gohtraw, 2025-04-08:
Silence Alert reappears after going to "POD MGMT" and tapping "Play Test Beep"

olorinmaia, 2025-04-08:
I don't experience this in android 12 and Samsung galaxy s10+

vanelsberg, 2025-04-09:
Interesting: will retry reproducing on my S22 (android 14)

vanelsberg, 2025-04-09:
Think I found the problem. Working on a fix - soon.

closed 2025-04-10 by MilosKozak