nightscout/AndroidAPS #3932: Cannot cancel Bolus if I switch to a different app and switch back

congruidavidli, 2025-04-04:
Reporting bugs
--------------
Repro:
1. Bolus any amount, for example 17 units
2. See that there is a stop button and a progress bar
3. Switch to another app
4. Switch back to AAPS
5. Now there is no way to top the bolus
6. There is a progress bar on the bottom, but no way to interact with it

- AAPS 3.3.1.3

vanelsberg, 2025-04-05:
When switching apps the pop-up progress window disappears.
Should not be a problem due to a fix giving user the option to cancel an ongoing bolus using the gray bar at the bottom ib Overview?

Confirming, this now also now is a **problem for 3.3.3.0-dev:** stopping congoing bolus can not be canceled using the gray bar in overview.

Btw: Progress is still on Wear Watch: it can be canceled there ok.

vanelsberg, 2025-04-05:
> * AAPS 3.3.1.3

Upgrade to latest release 3.3.2.0 and validate that canceling using the gray bar is possible.

olorinmaia, 2025-04-09:
I don't experience this problem on android 12 / samsung s10+ and latest dev 3.3.3.0

MilosKozak, 2026-10-11:
The bolus progress screen was rewritten in Compose (current dev version). It now always shows while a bolus is running, also after you come back from another app, and it has a Stop button. Please open a new issue if you still see this in 4.0.

closed 2026-10-11 by MilosKozak