nightscout/AndroidAPS #3935: Objective 4 Manual enact count not triggering as much as expected.

JoryAku, 2025-04-07:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem.

4th of April, I started objective 4 and noticed the count randomly stopped triggering after a few hours of using the temp targets. 

Downloading the user events csv I can see 37  events that are labeled TEMP TARGET. 
[2025-04-08_105631_UserEntry.csv](https://github.com/user-attachments/files/19640201/2025-04-08_105631_UserEntry.csv)

- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
AAPS 3.3.2.0

- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_

[Log Export.txt](https://github.com/user-attachments/files/19640222/Log.Export.txt)

  See https://wiki.aaps.app/en/latest/GettingHelp/AccessingLogFiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2025-04-09:
do you press press large button with recomendation to enact?

olorinmaia, 2025-04-09:
@MilosKozak I saw the discussion on Discord and I tested this myself. I've seen a few users thinking both enacting Open Loop suggestions AND Temp Targets should increase the counter. It's only Open Loop enacts that increase counter for manual enacts.

Here is the phrase for this objective:
Run in Open Loop mode for a few days and manually enact lots of temp basals. Set up and use temporary and default temporary targets (e.g. for activity or hypo treatment carbs)

Question is if only Open loop recommendations should be counted, if we could improve the string above. Else include Temp Targets in the counter as well, and possibly increase it?

JoryAku, 2025-04-09:
Oh I see, 
Yeah the wording made it seem like I was practising using the temp targets. 
I have been pressing the long button when it comes up. But it doesn't come up often.
I've only gotten 7 recommendations in the week or so I have been in this phase?

JoryAku, 2025-04-09:
@olorinmaia is there a setting to fire these recommendations off more? 
Or do I have to make my the space between my max and min targets narrower?
I am not getting very many recommendations.

olorinmaia, 2025-04-10:
I'm curious what @MilosKozak thinks about this objective. If it's meant to count only open loop enacts or also temp targets. For now I wouldn't do anything and leave the issue open. 

In loop settings there is % you can adjust for how often you will get suggestions. You also if you want manually set a 0 TBR and then trigger loop run. If you are not going low it will probably trigger a open loop suggestion.

MilosKozak, 2025-04-10:
The goal is to understand recomendations. If you have better wording you can create PR for discussion.
But it's about "accepting" and trying to understand