nightscout/AndroidAPS #3948: Question about the calculation of ZTpredBG:

doudou-ux, 2025-04-16:
[image]

As shown in the diagram, in the glucose prediction module, there are four predictions: IOB, COB, UAM, and ZT. 
Why is it that only ZT does not calculate predDev? 
I don’t quite understand the reasoning behind this design and would like to ask for your insights.

MilosKozak, 2025-04-16:
@scottleibrand

scottleibrand, 2025-04-17:
predDev is a short-term inertia term, which allows all the other prediction lines to start out pointing in roughly the same direction as the historical trendline, whether that's rising from carb absorption, dropping fast from something unexpected, or something in between. The ZTpredBG isn't intended to capture such short-term inertia: it's intended to visually show how much BG "should" be dropping from predicted insulin activity alone (if carb absorption were to suddenly cease), and predicts how the BG would trend if carb impact were to remain at zero, BG were to move according to predicted insulin impact, and the pump were to remain zero temped until BGs start rising above a threshold.

doudou-ux, 2025-04-17:
Thank you very much for your response. As I continued to think about the subsequent code, I only considered the aspect of being unaffected by carbohydrates. However, your explanation is much deeper and more comprehensive.

closed 2025-04-17 by doudou-ux