nightscout/AndroidAPS #3951: When Does minPredBG > maxCOBpredBG Occur? Seeking Examples or Diagrams for Clarity

doudou-ux, 2025-04-17:
[image]

I am very sorry to bother everyone again, but it’s really been quite challenging to understand, and I’ve been struggling to figure it out. 

Here’s the question: 

Under what circumstances would minPredBG > maxCOBpredBG occur, causing  maxCOBpredBG to be assigned to minPredBG ? Although I see there are comments in the code, I can’t seem to think of any scenarios. 

Could you provide an example or perhaps draw a diagram to explain? Looking for your reply！！！

Philoul, 2025-04-21:
Not sure it's the right place to ask this question 🤔.
This is core OpenAPS SMB algo wrote 8 or 9 years ago within oref project.

[image]

MilosKozak, 2026-10-11:
This is a question about the oref algorithm, not an AAPS bug, and it was answered above. For more detail, please ask in the oref0 project or on Discord.

closed 2026-10-11 by MilosKozak