nightscout/AndroidAPS #3954: I would like to see Google Health Connect integration

FreDiabetics, 2025-04-19:
Google Health connect could be a nice enhancement in future AndroidAPS versions.

https://developer.android.com/health-and-fitness/guides/health-connect/develop/get-started?hl=en

Gohtraw, 2025-04-27:
Why you ask?
It offers a convenient way to work with health data from different sources.
Juggluco can already send BG to it (I don't know how useful that detour would be, though).
Heart-rate and steps are already being implemented in AAPS - I think Health Connect could help streamline that process.
From what I've read, I'm sure connecting a period tracker to AAPS would be a really good idea.

I picked some example permissions which are self-explanatory; detailed info can be found here:  https://developer.android.com/reference/kotlin/androidx/health/connect/client/records/package-summary#classes

    <uses-permission android:name="android.permission.health.READ_ACTIVE_CALORIES_BURNED"/>
    <uses-permission android:name="android.permission.health.READ_BLOOD_GLUCOSE"/>
    <uses-permission android:name="android.permission.health.WRITE_BLOOD_GLUCOSE"/>
    <uses-permission android:name="android.permission.health.READ_DISTANCE"/>
    <uses-permission android:name="android.permission.health.READ_ELEVATION_GAINED"/>
    <uses-permission android:name="android.permission.health.READ_EXERCISE"/>
    <uses-permission android:name="android.permission.health.READ_FLOORS_CLIMBED"/>
    <uses-permission android:name="android.permission.health.READ_HEART_RATE"/>
    <uses-permission android:name="android.permission.health.READ_MENSTRUATION"/>
    <uses-permission android:name="android.permission.health.READ_STEPS"/>
    <uses-permission android:name="android.permission.health.READ_WEIGHT"/>

jhaydraude, 2026-01-14:
@FreDiabetics Are you thinking AAPS reads from Health Connect? Or that it writes to Health Connect?

In my exploration, I have found that (on my TicWatch 5)  AAPS's Wear app is a more reliable provider of step and heartrate data than Mobvoi is. I am currently implementing **Write** of this data to both Health Connect and Nightscout.

If AAPS as a data provider to Health Connect is the goal of this request, I can share my work.

alfureu, 2026-08-21:
Another idea is that I started using [SparkyFitness](https://github.com/CodeWithCJ/SparkyFitness), which has a pretty decent food input frontend (OCR, barcode import, etc.). It integrates Garmin data through Health Connect, but most importantly it also writes calories and nutrients to Health Connect.

This data can be tentatively used for bolus calculations... as suggestion with the possibility of adjusting etc. Just an idea.