nightscout/AndroidAPS #3960: Dexcom G7 Canada

Yosefrottenstein, 2025-04-24:
Hello
I am a proud user of the BYOD In Quebec Canada.
I have used it for the G6 app, on my small Unihertz Jelly 2 phone for many years.
However, my doctor recently told me that we have to switch to the G7 soon.
I was very disappointed that the BYOD supports G7 only the USA Version of the app, and not the Canadian version of the app. I have to use the Canadian version not so far for the mmol/l units. but mainly because my doctor is following me on the Canadian CLARITY platform.
I would strongly recommend that a Canadian, and/or International app version should be developed for the G7. It will be a great relief for many of us.
Thank you in advance for taking my words into consideration. I would like to hear back from you.

andyrozman, 2025-04-25:
I know this is major problem for you, but you are so to speak "barking at the wrong tree". AAPS has nothing directly to do with BYODA. While a lot of our users use it, it has nothing directly to do with us.

If you would ask on AAPS Facebook Group someone might point to the guy/team that is responsible for it, but we are not it...

MilosKozak, 2025-04-27:
Use FB page to find best solution

closed 2025-04-27 by MilosKozak