nightscout/AndroidAPS #3976: Proposed change: max carbs absorption limit in preferences

vanelsberg, 2025-05-08:
**AndroidAPS 3.3-dev**: Lates dev builds
**Flavor**: fullrelease

**Observation:**

I have been struggling with AAPS's carbs absorption for a long time. For me, carbs decay much faster than AAPS expects. In generally, this suggests IC/ISF profile settings may not be optimal.  

No matter what I try:  
- Either I have correct IC/ISF values, but carbs absorption in AAPS takes too long.  
- Or carbs absorption in AAPS is ok, but IC or ISF values are not correct.  

In both cases I end up with too much insulin and/or hypo's far too often.
In general, I solve this by either underestimating carbs intake or using latest option to enter negative carbs.  

Tried with higher carbs absorption settings in preferences. Maximum value for `min_5m_carbsimpact` is currently limited to 12 mg/dL/5 min.  A value of approximately 16 mg/dL/5 min proves to works much better for me.  

**Proposed change:**
Proposing to change the max value in setting min_5m_carbsimpact to 24 mg/dl/5min.

Before doing a PR for this change, I would appreciate feedback on this proposal.

Philoul, 2025-05-08:
My idea on this topic is a bit different, to fix this kind of issue.
You currently have 2 parameters in absorption settings to limit COB with low dev bars:
[image]

min_5m_carbimpact which will define (with CSF or ISF/IC) a minimum slope for COB curve
Max duration (between 4 hours and 10 hours)

Currently min slope is only limited by min_5m_carbimpact and profile, and if you still have remaining COB after max duration, then COB is replaced by 0.
- so during the period of carbs absorption, slope was too low to arrive at max duration with 0 COB max...

So my idea was to calculate min slope for COB curve using both parameters:
- when slope calculated with min_5m_carbimpact leads to 0 COB before max duration, this parameter is used
-when min_5m_carbimpact leads to 0 COB after max duration, then minimum COB absorption is calculated with max duration.

I'm not sure my explanations are very clear. 🙄

Philoul, 2025-05-08:
With my proposal, I don't consider slope depends on max duration, I just try to make max duration safety settings and min carb absorption consistant together.

Carb absorption should mainly be calculated with deviation, not with min_5m_carbimpact nor max duration...

I don't like IC/ISF additional parameter. The only justification is to override current limit of min_5m_carbimpact parameter.

Philoul, 2025-05-09:
#3359 is already open on this topic

vanelsberg, 2025-05-09:
> [#3359](https://github.com/nightscout/AndroidAPS/issues/3359) is already open on this topic

Yes. And it is about a possible solution for the underlying problem, so preferred. 👍
Agree that proposed change in here fixes a problem that also needs solving at algo level.

And as you document in the PR:
_>> Carb absorption speed is not modified when BG value is constant between 2 BG readings and DEV is above min_5m_carbimpact (so with high insulin activity after the meal if BG value doesn't increase...)_

And that seems to be exactly my problem? I often hover at relatively stable BG (higher than target but relatively "flat") for over an hour while high on IOB and activity decreasing as expected: but carbs decay far too low.

For me min_5m_carbinpact =16..18 (experimentally over 4 weeks) solves 75% of my problems.
As mentioned, I can get carbs decay a bit faster through profile but then I consequently end up with either the wrong IC or ISF.

*Q*: I'd say the value for min_5m_carbinpact depends on the person. Why limit it to a value of max 12?

My proposal would be to change the current max (1 line of code) and then continue with PR3359 to improve the underlying logic.

szantos, 2025-07-14:
Long ago, I've made the same experience, that `min_5m_carbimpact` is too low for me.

Actually it was a security issue, as without a rise in BG after meal carbs didn't get used up. A lot of COB was floating around combined with lower BG levels and with the slightest rise in BG the loop thought now it all will absorb and bolused too much and I got low.

The value I use in preferences for years is `min_5m_carbimpact = 18`(!) But don't judge this value on its own, it only makes sense in combination with my other settings. While tuning IC and ISF to my needs, I've always kept an eye on the calculated min hourly carb decay rate and don't let it go under 20 g/h.

`MinHourlyCarbDecayRate [g/h] = min_5m_carbimpact * I:C / ISF`

vanelsberg, 2025-11-16:
Btw: the current PR#4312 was ment as an intermediate fix for a more final solution as proposed by @Philoul in [issue#3359](https://github.com/nightscout/AndroidAPS/issues/3359)