nightscout/AndroidAPS #3977: Issue with BG Trend Arrow When Using Exponential Smoothing and boradcast data to GlugoDataHandler

AhmedSalehz, 2025-05-08:
Hello,

I’m currently using GlucoDataHandler to handle alarms and notifications. I have configured AndriosAPS as the local data source and set up AAPSClient to share data via Samsung Tizen. The data transmission is functioning as expected.

However, when the main smoothing method in AAPS is set to Exponential, GlucoDataHandler consistently shows a question mark instead of the BG trend arrow.

Note: The primary AAPS instance uses Juggluco as the data source, with BG data updated every minute.

Best regards,
Ahmed Saleh

reijoahola-nightscout, 2025-05-08:
May be this is somehow fixed already in GDH. I have v. 2.0 Beta release in use and ?-mark is not shown after a short test. I have Exponentoal filtering now on and follow this some hours in order to verify that it is fixed.

Note also that that GDH has a setting "Trend calculation" and it turn on internal calculation for Trend arrow. I have it now turned off for testing this issue.

reijoahola-nightscout, 2025-05-08:
GDH developer explained why GDH v. 2.0 looks like "fixing" the problem. GDH automatically turn on "Internal Trend calculation" if the received data does not contain trend info. There is also a manual setting for "Internal Trend Calculation" but it works also automatically.