nightscout/AndroidAPS #3978: A constant, unrelenting queue in xdrip

Inkalika, 2025-05-08:
I have sugar from juggluco going to aaps, and from it together with therapy to xdrip. Almost every day the therapy queue in xdrip does not decrease and cyclically overloads, and aaps starts to slow down. This goes on and on until i reboot the phone many times. Cleaned aaps cache, xdrip cache, cleaned base in aaps, disabled therapy transfer from aaps xdrip - all this does not help. The only thing left is to completely clean both applications and set up again. Maybe someone knows an algorithm to eliminate the queue of 200000 values from aaps xdrip?
Libra2 sensor, 1 minute readings, aaps version 3.3

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Inkalika, 2025-05-14:
> do database cleanup. if this doesn't help you must clear the database

Yeah, database cleanup doesn't work. Clear the database helps, but then the information about battery life, cartridge usage time is deleted, and the therapy in xdrip, which was in the queue, will not be transferred. I thought maybe there is some way to reset only the queue in xdrip without touching everything else

MilosKozak, 2025-09-25:
there is not