nightscout/AndroidAPS #3980: Feature Request: Improve glucose balancing under extreme weather conditions

MiryamMann, 2025-05-11:
Many people with type 1 diabetes experience noticeable changes in blood glucose levels during extreme weather, particularly during hot or freezing conditions.
I’d like to propose a new feature that integrates weather data (e.g., via an official meteorological API) into the balancing logic, so that environmental conditions can be taken into account dynamically.
This could help improve accuracy and safety for users during unusual climate events.

Let me know if this idea aligns with the project’s direction — I’d be happy to discuss possible implementation strategies.

andyrozman, 2025-05-11:
Theoretically connection to any weather API shouldn't be a problem, but each user would need to use its own credentials, because most of this services are not free... 
Second problem I see that none of this weather changes are quantifiable... So you can't say that every person will need x times more insulin if weather changes from one day to another by y degrees... Not everybody has same requirements... Some people need more insulin and some people need less insulin if weather changes and not even they have the same amount all the time... Even collecting such data could be problematic (even if you do it on yourself), even for initial collection you would need to exactly record all the intakes and not leave anything out... 
While idea seems intriguing, it would be hard to make and it wouldn't benefit that many people... 
If you wanted to create learning model (sort of AI), you would need to collect a lot of data (from different users even), and any invalid data could skew the model (for example any data from me would be totally useless, because I don't record CH and in most of cases I let Loop run on its own, without even giving additional boluses)...

MiryamMann, 2025-05-12:
Thank you for replying. 
I'll try to explain myself better. 
The way I thought to do this is as follows:
The program would of course be optional for every user. There would be a setting option to raise the bolus ceiling on hot days — meaning, the loop would have more freedom to correct and balance high blood sugar levels.
The percentage increase in the ceiling would be determined by the user, through self-learning and self-research. For example, I would personally allow a margin of 15%.

Regarding the cost issue — I think that in my country (Israel) it is possible to use the API for free, and probably in other places too. If needed, I will conduct more serious research on this.
Does my idea sound more reasonable now?

TestManMars, 2025-05-12:
I agree with @MiryamMann but I think the implementation should be different. 

A [research article](https://www.sciencedirect.com/science/article/pii/S2212877824000322) from 1981 shows the following:
[image]

How I think it can be done:
- User creates API key for free weather reports (e.g. https://openweathermap.org/api/one-call-3).
- The characteristics of the weather can optionally be used as triggers for automations (e.g. profile%, TT, etc.)

@andyrozman 
- "Not everybody has same requirements... Some people need more insulin and some people need less insulin if weather changes and not even they have the same amount all the time..." => We can leave this up to the user to decide right? For me setting up an automatic profile change to 60-80% would already decrease my mental workload and the chance of an unexpected hypo . 

- If you wanted to create learning model (sort of AI), you would need to collect a lot of data (from different users even), and any invalid data could skew the model (for example any data from me would be totally useless, because I don't record CH and in most of cases I let Loop run on its own, without even giving additional boluses)... => I think AI is indeed a lot of effort and will be not more reliable than a self defined automation rule.

MiryamMann, 2025-05-12:
Understood. Thank you for the time you took to write your response. I’d be
happy to help in other ways. If you have an idea for a task I could take
on—to contribute or improve something—I’d be glad to hear

‫בתאריך יום ב׳, 12 במאי 2025 ב-13:12 מאת ‪TestManMars‬‏ <‪
***@***.***‬‏>:‬

>
> I agree with @MiryamMann <https://github.com/MiryamMann> but I think the
> implementation should be different.
>
> A research article
> <https://www.sciencedirect.com/science/article/pii/S2212877824000322>
> from 1981 shows the following:
> Screenshot.2025-05-12.at.10.01.02.png (view on web)
>
> How I think it can be done:
>
>    - User creates API key for free weather reports (e.g.
>    https://openweathermap.org/api/one-call-3).
>    - The characteristics of the weather can optionally be used as
>    triggers for automations (e.g. profile%, TT, etc.)
>
> @andyrozman <https://github.com/andyrozman>
>
>    -
>
>    "Not everybody has same requirements... Some people need more insulin
>    and some people need less insulin if weather changes and not even they have
>    the same amount all the time..." => We can leave this up to the user to
>    decide right? For me setting up an automatic profile change to 60-80% would
>    already decrease my mental workload and the chance of an unexpected hypo .
>    -
>
>    If you wanted to create learning model (sort of AI), you would need to
>    collect a lot of data (from different users even), and any invalid data
>    could skew the model (for example any data from me would be totally
>    useless, because I don't record CH and in most of cases I let Loop run on
>    its own, without even giving additional boluses)... => I think AI is indeed
>    a lot of effort and will be not more reliable than a self defined
>    automation rule.