nightscout/AndroidAPS #3984: AAPS dev 3.3.3.0 froze

robertrub, 2025-05-16:
This happened several times. I look at AAPS and find out that is has frozen during the night.
I thought that it was related to Play Protect but it froze again this morning (a bit before 7 AM) and Play protect is OFF.

As you can see, there was a "catch-up" in BGs when I force closed, cleared cache and restarted AAPS.

[image]

The logs after force close and restart.

[image]

TheNeskik, 2025-05-16:
Got an other freeze to this night.

Since last time I enabled everything to log as I don't know what can cause those freeze.

[image]

[image]

I just updated to latest dev this morning after the freeze, I will see if it happen again

MilosKozak, 2025-05-17:
> This happened several times. I look at AAPS and find out that is has frozen during the night. I thought that it was related to Play Protect but it froze again this morning (a bit before 7 AM) and Play protect is OFF.
No reason is visible in logs

WesSec, 2025-05-17:
I experienced the same 2 times on (86fb). Checked the logs and also no reason visible. It didnt crash, it just hung on the latest sate. (it showed last bg was 3 min ago, so upon checking first i didnt notice it was broken). I pulled latest commits and see if i can get a logcat dump when it happens again:

Extras:
Phone: Samsung S21FE, OneUI 6.1 (Android 14)

TheNeskik, 2025-05-17:
It happened on older commit (f737) but nothing in log. Can't replicate issue as it is random.

I'm on Android 15 (galaxy S23)

robertrub, 2025-05-17:
`It didnt crash, it just hung on the latest state.`

Same here. No crash, just froze.  No reaction to taps etc.

vanelsberg, 2025-05-17:
> Same here. No crash, just froze. No reaction to taps etc.
Only AAPS freezes - other apps and phone are then still responsive?

TheNeskik, 2025-05-17:
Only AAPS,
I'm using byoda and it still get BG as usually.
Phone can be use as usual

Rabuffulin, 2025-05-20:
o meu AAPS 3.3.2.0 também trava. O comportamento dele é parecido com o LOOP Aberto. A quantidade de hidratos informada não diminui e as correções não acontecem. O problema é aleatório. Essa noite ele não fez uma única correção e amanheci com a glicemia alta, com 60g de hidratos que foram do meu jantar de ontem. Estava ativa somente a basal programada. Quando fui na aba SMB e atualizei ele gerou uma correção e aplicou, mas se não tivesse feito isso permaneceria congelado. Agora tomei meu café da manhã e parece que está funcionando o LOOP Fechado, mas se isso ficar acontecendo eu pensei em fazer um downgrade de versão, porque na 3.3.1 isso não acontecia.

vanelsberg, 2025-05-21:
> o meu AAPS 3.3.2.0 também trava. O comportamento dele é parecido com o LOOP Aberto. A quantidade de hidratos....

Here's the translation into English:

"My AAPS 3.3.2.0 also freezes. Its behavior is similar to Open Loop. The amount of carbs entered does not decrease, and corrections do not happen. The problem is random. Last night, it didn’t make a single correction, and I woke up with high blood sugar, with 60g of carbs remaining from my dinner yesterday. Only the programmed basal was active. When I went to the SMB tab and refreshed it, it generated a correction and applied it, but if I hadn’t done that, it would have remained frozen. Now I’ve had breakfast, and it looks like Closed Loop is working, but if this keeps happening, I’m considering downgrading the version because this didn’t happen in 3.3.1."

Hope that helps! Let me know if you need any refinements.

Rabuffulin, 2025-05-21:
My AAPS 3.3.2.0 also freezes. Its behavior is similar to Open Loop. The amount of carbs entered does not decrease, and corrections do not happen. The problem is random. Last night, it didn’t make a single correction, and I woke up with high blood sugar, with 60g of carbs remaining from my dinner yesterday. Only the programmed basal was active. When I went to the SMB tab and refreshed it, it generated a correction and applied it, but if I hadn’t done that, it would have remained frozen. Now I’ve had breakfast, and it looks like Closed Loop is working, but if this keeps happening, I’m considering downgrading the version because this didn’t happen in 3.3.1

WesSec, 2025-05-23:
>  xDrip had gotten all the BGs but they did NOT catch-up when AAPS was "force restarted".  

This happened for me too, i went to the shower and when i got back (phone was in range all the time), AAPS started receiving BG's again. Here's my reconstruction

- Freeze around 4:17
- Discovered around 6:50 (got pop upAAPS froze, force closed it and restarted AAPS
<no bg's being received AFTER restart>
- BG's received since 07:08 (approx 20 min after restart)

Logcat logs mentioning AAPS (unfortunately crash is out of retention period)

[froze_redacted_aaps_log.txt](https://github.com/user-attachments/files/20405282/froze_redacted_aaps_log.txt)
[logcat_filtered_aaps.txt](https://github.com/user-attachments/files/20405283/logcat_filtered_aaps.txt)

[image]

vanelsberg, 2025-05-23:
_Using BYODA and AAPS latest dev versions.
Not experiencing "freezes" and It seems to me problems reported most are with xDrip+_

My 2cts:
(1) It seems to mee Google has been making Play Protect behaving more and more aggressive.
Since my phone received an OS update month ago, it started to "block" AAPS. I can unblock, but (Samsungs) phones security keeps reminding about it being a "security risk"

(2) _What is reported in this issue looks similar to what I experienced with BYODA some weeks ago_ after my Samsung S22 Ultra (running A14) received an OS update where Play Protect suddenly started to behave even much more aggressive. Until then I never had a need to switch it off.

Shortly after, on updating AAPS, Play Protect threw a big pop-up. Ignored it and installed AAPS anyway. That night I experienced with AAPS which has a striking resemblance with the "freezing" reported here: No problem with BYODA/Dexcom which was receiving BG values as usual, but AAPS just stopped receiving BG values from BYODA and loop stopped. Could not find the problem so got through the night with just profile basals.

Next morning it took me at least an hour to solve the problem:
- AAPS did not show any problems other than not receiving BG from BYODA.
- Switching of Play Protect and any power savings still active did not solve.
Still, my idea was this could have do with permissions, but all permissions seemed OK ?!?

Then I noticed under AAPS permissions, "additional permissions" it now showed the "Access Dexcom app".
Allowing/adding this permission solved the problem and AAPS immediately started receiving BG values!
Noticed that the Dexcom app (BYODA) now also shows itself under "App Additional permissions" which I find a bit odd?

@MilosKozak 
I do have xDrip+ on the phone as a companion app, only receiving BG's from AAPS. For AAPS there is no such permission for xDrip+ under "App Additional permissions". Could it be AAPS needs to set some additional permissions similar to BYODA?

robertrub, 2025-05-23:
Strange, I had that "App Additional permissions" line in AAPS permissions but when I checked right now, I can't see it.

It was there for some time now and had receive data from xDrip (os sth similar)

TheNeskik, 2025-05-23:
I'm using BYODA and i'm having same behavior randomly.
What i've made since my last freeze :

- Updating AAPS
- Reset permissions and giving all access again
- Reinstall BYODA (to be sure and because last freeze was giving me trouble to get bg again in aaps (around 15 min after restart of aaps and then all bg were backfilled))
- Check again if play protect is off (it was)

About Additionnal Persmissions it was a while since it is here (can't remember when I saw it) but it was already on for me.

Using :
Samsung S23 - Android 15
AAPS-dev- e266
Not using dynisf nor autoisf nor autosens.

WesSec, 2025-05-24:
I'm in a freeze state right now, and am watching logcat for androidAPS, while the app hangs, i see a lot of these warnings pass by:

```
2025-05-24 17:55:54.157 28348-24686 SQLiteConnectionPool    info.nightscout.androidaps           W  The connection pool for database '/data/user/0/info.nightscout.androidaps/databases/androidaps.db' has been unable to grant a connection to thread 1368 (RxCachedThreadScheduler-277) with flags 0x2 for 1757.128 seconds.
                                                                                                    Connections: 0 active, 1 idle, 3 available.
2025-05-24 17:55:54.158 28348-28971 SQLiteConnectionPool    info.nightscout.androidaps           W  The connection pool for database '/data/user/0/info.nightscout.androidaps/databases/androidaps.db' has been unable to grant a connection to thread 181 (arch_disk_io_2) with flags 0x2 for 1757.129 seconds.
                                                                                                    Connections: 0 active, 1 idle, 3 available.
2025-05-24 17:55:54.159 28348-28988 SQLiteConnectionPool    info.nightscout.androidaps           W  The connection pool for database '/data/user/0/info.nightscout.androidaps/databases/androidaps.db' has been unable to grant a connection to thread 184 (arch_disk_io_3) with flags 0x2 for 1757.1171 seconds.
                                                                                                    Connections: 0 active, 1 idle, 3 available.
2025-05-24 17:55:55.019 28348-5570  SQLiteConnectionPool    info.nightscout.androidaps           W  The connection pool for database '/data/user/0/info.nightscout.androidaps/databases/androidaps.db' has been unable to grant a connection to thread 1510 (RxCachedThreadScheduler-290) with flags 0x2 for 1721.1881 seconds.
                                                                                                    Connections: 0 active, 1 idle, 3 available.
2025-05-24 17:55:56.952 28348-26280 SQLiteConnectionPool    info.nightscout.androidaps           W  The connection pool for database '/data/user/0/info.nightscout.androidaps/databases/androidaps.db' has been unable to grant a connection to thread 1381 (RxCachedThreadScheduler-278) with flags 0x2 for 1733.1411 seconds.
                                                                                                    Connections: 0 active, 1 idle, 3 available.
```

(i'll update this post as i try to debug whats happening)

robertrub, 2025-05-26:
@MilosKozak I had not done any full sync ( but my phone only uploads to NS (api v3) when charging at night) but had the freezes too.

Didn't have any freezes for some days. Running dev built on 18/05.

vanelsberg, 2025-05-26:
Update:... just realized *I have been merging PR3968*: 
@MilosKozak _With that I do not experience any freezing._

Just installed current AAPS dev commit _without_ merging PR3968 on my Nokia 5.4 test phone: it completely freezes (close app / wait pop-up) when trying a full NS sync....  When I then "close" AAPS, reopen I can continue full sync successfully. Subsequent sync are experiencing som app freezing but eventually finish successfully also.

Also noted during full sync queue is not used (at least UI shows "queue: n.a")

So PR3968 fixes a problem that is in current dev?

WesSec, 2025-05-27:
> > Solved my issue, in the database was a carbs entry with a duration of 216000000000 (default for unlimited).
> 
> Good find! Question is, how did it get there ;-)

Been able to reproduce the issue now

[image]

My steps:

- Clear database
- Attempt full sync
- Switch to action tab during sync
- App freezes for me with following logcat messages:
[image]
- Quit app as suggested by prompt
- Start app again
- This behavior likely corrupts/interrupts the database entry
Proof that the mongo entry is okay:

[image]

Solving this issue completely would likely require an overhaul of the database entry / sync mechanism to catch these occurrences, suggested fix in #4002 is a solution to prevent the app from crashing if it happens

robertrub, 2025-05-29:
When I look in the treatments tab, the only thing near 6:25 is this. And it is a bit strange, 14 min not 15 min !temp basal.

[image]

robertrub, 2025-06-08:
Hi @vanelsberg 
No, I can't trigger the freeze. It happens all by itself ;)
I use (recent) nightly xDrip, BG sent to AAPS only, AAPS uploads everything to NS only. Using AAPS _dev_ , rarely older than 7-10 days.
I didn't have any freeze problems for around 7-10 days.
I tried upload data to xDrip for an hour only, in order to see what data looked like in xDrip. Worked OK.

vanelsberg, 2025-06-08:
Hi @robertrub 
> No, I can't trigger the freeze.

Ok. I think the "freezing" you were reporting is a different one then mine:

(1) this morning installed latest dev without merging the StateDB PR on my main phone. This required a database reset as I had been running with the PR for some time. Half way resyncing data from NS, AAPS looked up with the wait/close pop-up. Waiting did not solve so exit AAPS. On restart it was fine, so after letting it stabilize for a few minutes tried a full resync again. 2nd resync completed where it left off successfully.
Will open a new issue for this,

(2) Apart from above, I did experience a "freeze" that is similar to what you described twice:
AAPS suddenly stopped receiving BG values causing loop to stop doing its work. As far as I could see somehow BYODA (which was registering BG values as expected) was somehow blocked from sending BG values to AAPS. Think this is an OS issue, suspecting Google's Play Protect updating causing this.

robertrub, 2025-06-08:
```
AAPS suddenly stopped receiving BG values causing loop to stop doing its work. As far as I could see somehow BYODA was ... somehow blocked from sending BG values to AAPS.
```

In my several freeze cases, AAPS didn't just stop receiving BG values, it also did NOT react to any taps etc. The UI (or the full app) was dead or frozen.

vanelsberg, 2025-06-08:
> ...The UI (or the full app) was dead or frozen.

OK, so now to me this gets complicated with 3 different "freeze" situations 😕
1) AAPS freezing, not UI not responding.
2) AAPS Freezing (wait/exit) while doing a full NSv3 resync.
3) AAPS "freezing" (in loop stopping), not receiving BG values.

Now question is, are these different, or maybe somehow related?

robertrub, 2025-06-13:
More info. Looks like the last entry in the treatments is an lower profile% automation but it might have resulted in an unsuccessful pump communication (pump unreachable alert screen I saw for a fraction)
[image]

[image]

[image]

Phone Android 15 Samsung OneUI 7.0 and got a minor (security ?) update yesterday. Phone was on charger, I was sleeping. I woke to the xDrip hypo alert ans saw the frozen AAPS.

robertrub, 2025-06-13:
Last screenshot. What NSV3 looks like
[image]

TheNeskik, 2025-06-13:
One more thing on my side and what 
I don't experience freeze anymore since my last comment.

But I see that Play protect is very aggressive since A15. Even when update AAPS, it's trying to re-enable itself (happened to me yesterday, after update Play protect tried to come back)
So be careful that it didn't activate again.

I was using Byoda and I had those freeze. Maybe Play protect ? Since A15 I had an other permissions to AAPS to access BG from Byoda. I know that Android is trying to clean permission from app that are not used from a certain time.

I'm currently using juggloco as I switch from g6 to one+ sensor and I didn't experience freeze anymore.

But I highly suspect A15 behind those issue.

As it was already said, I think we need to separate different freeze issue. As I think they are not the same and probably different issue

robertrub, 2025-06-13:
```
At 00:31:28 I see an error concerning profile:
00:31:29.032 [main] D/CORE: [AlarmSoundServiceHelper.startAlarm():34]: Starting alarm from La mise à jour du profil basal a échouée
```
That's when the Automation tried changing my profile to 30% but couldn't succeed.

That's why @MilosKozak thinks that it might be a driver problem but something else can also stop the driver talking to the pod....

robertrub, 2025-06-13:
```
Maybe Play protect ?
```
Nope, first thing I checked after the freeze. It was still OFF.
Yes, A15 proposes to activate Play Protect each time I update AAPS...

vanelsberg, 2025-06-13:
@robertrub  Had a freeze that looks just like you have been describing.

Q: What connections settings do you use for NSv3 (WiFi/Cellular)?
My connection settings are set to allow both WiFi (home) and Cellular. But what happens when there is no WiFi and Cellular coverage is low?

Today I think I experienced an identical "freeze" while being away from home, diving my car in an area were Cellular connect might be intermitted. When home I tried opening AAPS and got a white screen with logo, no more. No wait/close. Watch showed loop had not been running for 50 minutes (so approx. stopped while I was away).
Trying to find what was wrong AAPS suddenly started working again. Al BG's were, likely because of BYODA backfilling

Checked my logs and noticed a large number of "NETCHANGE: Mobile connected" messages. When searching for "**wifiConnected=false**" I found  a list of "**[NetworkChangeReceiverHandler]** ... NETCHANGE: Mobile connected" events. These events log entries started at about the time I left Home until I got Home again.

So, I decided to recheck _your_ logfiles:
Searching for "wifiConnected=false" and "NetworkChangeReceiverHandler" showed identical logging from 00:23:41 until 01:11:29 when your APPS froze. Also noticed similar logging for 2025-06-12 but for much shorter periods (18:53..19:10 and 22:01:34...22:01:36) so you may not have noticed them.

Expect I should be able to repro by simulating no WiFi, no Cellular.
Speculating on possible cause of the problem: AAPS might not handle network disconnect and keeps trying. Maybe some memory leak problem, not closing threads properly or cleaning unused resources? To find out I need a repro and logcat as logging does not show.

jbr7rr, 2025-06-13:
> [@jbr7rr](https://github.com/jbr7rr) this seems to be Medtrum pump or driver issue

@MilosKozak Not sure, reading back I also see at least one Dash user reporting this.
Anyway I will update to latest dev and see if I also run into this issue and if I can replicate from there

robertrub, 2025-06-13:
@jbr7rr Samsung S21 FE 5G, Android 15, OneUI 7.0. 
AAPS dev. G6, Anubis, xDrip daily (recent). BGs from xDrip to AAPS and then to NSV3. Upload while charging.
Medtrum Nano

robertrub, 2025-06-13:
> Q: I also have set "Slow down uploads" option set to "ON"...

Why on earth should I slow uploads ? I have great WiFi, NS is on my NAS which is mighty fast too. 

Why do you think this is WiFi or upload related ? The phone has been charging and uploading regularly for over 3 hours before AAPS froze. There was no upload queue, no bottlenecks, no upload problems.

vanelsberg, 2025-06-14:
> Why do you think this is WiFi or upload related ? 

Just trying to find common ground, not so much because of upload.
However, your and my "freeze" logs have in common that wifi was down and no mobile connection.... somehow.

vanelsberg, 2025-06-14:
> Why on earth should I slow uploads

Not should, but no need not to. Even with slow, my NS is updated almost instantly on treatments.
I expect "slow" to prevents high app load and additional battery usage when NS things get busy?

robertrub, 2025-06-14:
```
your and my "freeze" logs have in common that wifi was down and no mobile connection.... somehow
```
That is very strange as the house is well covered by WiFi and 5G. I can't see why **both** would disappear. 🤔

vanelsberg, 2025-06-18:
**AndroidAPS 3.3.3.0-dev-a**
**Build**: 3133121c76-2025.06.11
**Flavor**: Fullrelease
**Phone**: Samsung S22 Ultra (Android 15)
**Pump**: Medtrum Nano U200
**CGM**: G6/BYODA
**NSv3**: Release version 15.0.3

**Problem description**:
AAPS UI unresponsive/frozen. Had to restart the App.

**Additional information:**
System was running stable, no problems.
Recently upgraded the phone from Android 14/OneUI6 to 15/OneUI7 

This looks exactly as @robertrub has been describing:
Freezing does not seem to relate to the freezing I reported earlier (which looked more like OS blocking resources).
This time UI was completely unresponsive presenting the Wait/Close pop-up.

**Repro:**
Not found

**Configuration:**
Sensitivity Oref1, OpenAPS SMB with autosense

**Log files:**
No clues found on this problem. App seemed to be doing work in the backgrand?
 
[AndroidAPS_LOG_1750209886144.log.zip](https://github.com/user-attachments/files/20797028/AndroidAPS_LOG_1750209886144.log.zip)

**Screenshots:**

Frozen:
[image]

On restart/init:
[image]
[image]

vanelsberg, 2025-06-18:
> ... difference is that you had your BGs fill up after

Using BYODA. According to logs, AAPS still was processing BG values to NS in the background?

robertrub, 2025-06-18:
```
processing BG values from BYODA to NS
```
Ok, the difference is that I'm using xDrip. The BGs were missing from AAPS and NS too (but I'm not 100% sure about NS now...)

robertrub, 2025-07-12:
Frozed again. xDrip has all the BGs. AAPS doesn't react to anything. Android shows the message "App not working. Wait or close ?". Chose "Close" and then restarted AAPS. Last BG in AAPS at 2:26. Logs exported and screenshots after restart of AAPS. Can't recall if the graph showed till 5AM (actual time) before restart...

[image]

[image]

[image]

robertrub, 2025-07-12:
Frozed again. xDrip has all the BGs. AAPS doesn't react to anything. Android shows the message "App not working. Wait or close ?". Chose "Close" and then restarted AAPS. Last BG in AAPS at 2:26. Logs exported and screenshots after restart of AAPS. Can't recall if the graph showed till 5AM (actual time) before restart...

[image]

[image]

[image]

EDIT.... Very strange. After close by Android _wait/close?_ message and after 10 minutes, still no BGs in AAPS... I got an alert from AAPS saying BGs missing. xDrip still getting BGs. Force closed and cleared cache memory of AAPS and restarted abd still no new BGs in AAPS.

[image]

New logs

[image]

Question : I have DuckDuckGo stopping outgoing tracking "messages". This stops AAPS sending a lot of tracking data to Google too. Can this be a reason to the freezes ? I don't think so as AAPS is working normally most of the time but who knows...

[image]

TheNeskik, 2025-07-12:
I had a freeze yesterday too with xdrip.
I had my phone in my hand so I saw it, only 10min freeze. Nothing in log.
As it happened with Byoda or xdrip it is not related to those app. 
Xdrip do not backfill data when AAPS crash.
I do not stop tracking from Google. So I would say it is not related to.

vanelsberg, 2025-07-21:
Looks like we are seeing different kinds of "freezing"? From what I have been experiencing:
1. Total lockup of AAPS fro some reason: (It get the feeling it relates setting profile % through automations?)
2. Loosing BT access on BYODA and pump like Medtrum & DASH: (OS permissions reset?)

This morning, I had another freeze while using the phone were BYODA's BG values were no longer received (so loop stopped). Trying to operate AAPS it started presenting the wait/close pop-ups. Even after restarting the phone.

This really looked like some OS problem with permissions getting lost on AAPS<->BYODA? As on previous occasions (I think) **Solved the problem by** by going through

- settings->app->AAPS->Permissions->Additional permissons: "Access Dexcom App".

It still was set to "Allow" but when I toggled it to "Don't allow" (and back) I got the OS popup on granting permission!?
After granting to problem solved and AAPS started running as expected!

robertrub, 2025-09-22:
I hit AAPS widget and I get a black or white screen, nothing happens. I try to get the main screen of the phone but usually the phone doesn't answer to the center button. I use the left button (show open apps) to change the forward app. Then, I can get the main screen. Then, it's easy and fast : force close AAPS, clear cache of AAPS, restart AAPS.

vanelsberg, 2025-09-22:
> NS is missing the values during the freeze period.

Same for me, except for the fact that at the moment AAPS unfreezes, it backfills BG's from the BYODA/Dexcom app.

robertrub, 2025-09-22:
```Wondering if your freezes and mine are of the same origin```

Mine happen (almost ?) at night, early morning.

Can't understand why it happens often after an AAPS update (but not really kept track of it). Play Protect is OFF

robertrub, 2025-09-25:
> "Highly" recommended setup is to let only AAPS write to NS

Yes, that's what I remember and have done. AAPS uploands all.

robertrub, 2025-09-30:
Mine frozed this morning around 5AM. I woke up just after the beginning and fir ed closed to restart it. See the missing BG in AAPS. There were some automations (profile changes against negative IOB) before the freeze but not just before it

[image]

[image]

[image]

[image]

robertrub, 2025-10-03:
Froze again. No automations,  nothing fancy before the freeze.
Using latest dev.
I wonder why @vanelsberg and I have these freezes but not many other users 🤔

[image]

[image]

robertrub, 2025-10-03:
In fact, AAPS is NOT totally frozen (I suppose) as xDrip goes on sending BGs and (probably) getting the _BG received_  acknowledgement (if there is such a thing...) xDrip considers the BGs as sent (and received ?) and doesn't sent them again when AAPS restarts and becomes reactive.

swissalpine, 2025-10-06:
Today a new freeze: BluetoothAdapter not initialized_ERROR

Log:
```
05:00:24.259 [DefaultDispatcher-worker-2] E/PUMPBTCOMM: [BLEComm.connectGatt():171]: connectGatt, mBluetoothGatt is not null
05:00:39.983 [DefaultDispatcher-worker-6] D/CORE: [LoadDeviceStatusWorker.doWorkAndLog():38]: DEVICESTATUSES: [NSDeviceStatus(app=AAPS, identifier=f973cbd2-6836-5707-b4db-a7c23ed3b072, srvCreated=1758769025050, srvModified=1758769025050, createdAt=2025-09-25T02:57:04.731Z, date=1758769024731, uploaderBattery=100, isCharging=true, device=openaps://motorola motorola razr 50, uploader=Uploader(battery=null), pump=Pump(clock=2025-09-25T02:57:04.732Z, reservoir=95.0, reservoirDisplayOverride=null, battery=Battery(percent=null, voltage=null), status=Status(status=normal, timestamp=2025-09-25T02:52:13.756Z), extended={"lastBolus":"25.09.25 04:52","lastBolusAmount":0.05,"TempBasalAbsoluteRate":0.9500000000000001,"TempBasalStart":"25.09.25 04:52","TempBasalRemaining":25,"BaseBasalRate":0.7000000000000001,"ActiveProfile":"Standard 25-09(110%)"}), openaps=OpenAps(suggested={"algorithm":"SMB","timestamp":"2025-09-25T02:56:59.712Z","bg":154,"tick":"-2","eventualBG":96,"targetBG":100,"insulinReq":0,"deliverAt":"2025-09-25T02:56:56.900Z","sensitivityRatio":1,"reason":"COB: 0, Dev: 3, BGI: -2, ISF: 54, CR: 12,27, Target: 100, minPredBG 111, minGuardBG 96, IOBpredBG 99, UAMpredBG 96; Eventual BG 96 < 100, setting 0.44U\/hr. ","duration":30,"rate":0.44000000000000006,"predBGs":{"IOB":[154,152,151,149,147,146,144,142,140,139,137,135,134,132,130,129,127,126,124,123,122,120,119,118,117,115,114,113,112,111,110,109,109,108,107,106,105,105,104,103,103,102,101,101,100,100,99],"ZT":[154,152,150,147,145,143,141,140,138,136,135,133,132,131,130,129,128,127,127,126,126,125,125,125,125,126,126,126,127,128,128],"UAM":[154,152,150,148,146,144,142,140,138,136,134,133,131,129,128,126,125,123,122,120,119,117,116,115,114,113,112,111,109,109,108,107,106,105,104,103,103,102,101,100,100,99,99,98,97,97,96]},"COB":0,"IOB":1.183,"variable_sens":0,"consoleLog":["ISF unchanged: 54.5","EventualBG is 96.0 ;","minPredBG: 111.0 minIOBPredBG: 98.0 minZTGuardBG: 125.0"," minUAMPredBG: 96.0"],"consoleError":["----------------------------------","Activity detection: ","----------------------------------","Activity detection disabled: Phone seems not to be carried for the last 15 m. ","----------------------------------","Sensitivity ratio set to 1.0 based on autosens; ","Basal unchanged: 0.7000000000000001; ","CR:12.272727272727272","SMB enabled due to enableSMB_always","profile.sens: 54.54545454545455, sens: 54.5, CSF: 4.4407407407407415","Carb Impact: 0.6 mg\/dL per 5m; CI Duration: 0.0 hours; remaining CI (~2h peak): 0.0 mg\/dL per 5m","UAM Impact: 0.6 mg\/dL per 5m; UAM Duration: 0.3 hours"," avgPredBG: 125.0 COB: 0.0 \/ 0.0","BG projected to remain above 100 for 230 minutes","naive_eventualBG: 90.0 bgUndershoot: -20.0 zeroTempDuration 240 zeroTempEffect: 153 carbsReq: -39"],"isfMgdlForCarbs":54.54545454545455}, enacted=null, iob={"iob":1.183,"basaliob":-0.615,"activity":0.0082,"time":"2025-09-25T02:56:59.712Z"}), configuration=Configuration(pump=null, version=null, insulin=null, aps=null, sensitivity=null, smoothing=null, insulinConfiguration=null, apsConfiguration=null, sensitivityConfiguration=null, overviewConfiguration=null, safetyConfiguration=null))]
05:00:58.838 [DefaultDispatcher-worker-2] E/PUMPBTCOMM: [BLEComm.connectGatt():171]: connectGatt, mBluetoothGatt is not null
05:01:02.383 [BLECommHandler] E/CORE: [BLEComm.handleNotInitialized():479]: BluetoothAdapter not initialized_ERROR
05:01:02.949 [BLECommHandler] E/CORE: [BLEComm.handleNotInitialized():479]: BluetoothAdapter not initialized_ERROR
05:36:27.122 [DefaultDispatcher-worker-7] E/CORE: [KeepAliveWorker.doWorkAndLog():125]: KeepAlive fail
```

vanelsberg, 2025-10-07:
> Today a new freeze: BluetoothAdapter not initialized_ERROR

The "freezes" I have been having where not related to Bluetooth: no BT related errors in the log, AAPS simply does not receive BG values and the AAPS process threads seem get locked so Android starts throwing the dreaded Close/Wait pop-ups.

When this happend, AAPS is still connected to the pump and Dexcom/BYODA is still processing BG values. To me it looks like a problem with AAPS/BYODA inter-process communications that is somehow blocking.

Unconfirmed:
- Feels like the problems occurs more frequently when Profile% automations are active?
- Think this is more an issue with Android OS (Samsung A15/OneUI 7.0)
- Only few users are seeing the problem (Me and @robertrub  are both running Samsung A15?)

vanelsberg, 2025-10-07:
> I'm using xDrip too in combination with Medtrum. It happens if I leave the room without my smartphone.

Is this actually a "freeze" (where AAPS gets unresponsive, Loop stops but pump is still connected) or are you reporting a lost BT connection to the pump that does not repair itself? If yes, this then is a totally different problem that requires its own Github issue?

HuismanErik, 2025-10-19:
Hi, last night I also experienced a hanging AAPS :(
After some digging I found this thread.
I’m currently using the Dash pump and xDrip as my BG source, and I’m running a dev build from last week. My phone runs on Android 14.

@vanelsberg  AAPS stopped receiving (or process) blood glucose values after I restarted my phone. AAPS started processing them once I switched the BG source to Nightscout and then back to xDrip. I’m wondering if it’s really necessary to reinstall AAPS in this case.

robertrub, 2025-10-19:
@HuismanErik No, no need to reinstall AAPS. I usually kill AAPS, empty cache memory and restart AAPS. Works all the time. Using Nano and xDrip on Android 15 phone.

vanelsberg, 2025-10-20:
Other (different) freeze today: after updating to latest dev and selecting SMB preferences, AAPS locked up. On restarting AAPS I only got a blank/white screen. This time, force-stop + clear cache solved.

(btw: @HuismanErik The general freezes I reported here do not solve by clear cache / restart: in moist cases resetting AAPS/BYODA permissions get BG processing going again. If not, UI eventually starts locking up and only way to solve is un/reinstall.)

HuismanErik, 2025-10-20:
I tested my fix. But it did not work.

vanelsberg, 2025-11-16:
With current dev I have not seen any freezes lately ....

closed 2025-11-23 by MilosKozak