nightscout/AndroidAPS #3986: Insulin peak time. "Activity" or "concentration"?

kbountro, 2025-05-17:
Hi, I am trying to understand the role of the insulin model. What I struggle with is that the description of the variables indicate that, for example Fiasp's peak activity is 55 minutes. However, if we define activity as the glucose decreasing process, this is not true. This time is supposedly the peak concentration of insulin in serum. It is AFTER that time that the insulin works the best. There are clinical studies that report some figures regarding this (I read 115 min is an average peak activity for Fiasp).

I know the model is (primarily?) used to compute IOB. Can someone explain to me the benefit of assuming peak time = peak concentration time for this computation?

Thank you!

Philoul, 2025-05-21:
Everything is within these 2 curves
[image]

When you inject 1 unit of insulin within your body at t0, the remaining insulin will follow the pink curve from 1U at t0 to 0U at t0+DIA hours.
The derivative of the pink curve as a function of time is the blue curve, and this blue curve is called "Insulin Activity"

In a nutshell, if your ISF value is constant (to simplify explains), with a value of 50 mg/dl /U, then if you deliver 1U of insulin, your BG value should be 50 mg/dl lower DIA hours later, and the variation of your BG value will follow the same "pink curve" from t0 to t0+DIA.

The Activity curve is the slope of the pink curve, and peak time is when the variation of BG value between 2 received value will be the most important.
Insulin activity units are U / min (or U / 5min if your sensor send values every 5 min), and the expected variation of your BG value between now and now+5min is calculated that way:
DeltaBG = InsulinActivity * ISF
So the highest expected variation of your BG value will be peak min after the bolus, whenthe blue curve is at the highest level.

kbountro, 2025-05-22:
Hi Philoul, thanks for your answer. To be honest, I don't think this is 100% correct, and it also does not answer my question. Let me be more precise.

- If the blue curve is the "Concentration" (let me use this term for now), then the "IOB" at time t is 1 minus the integral of "Concentration" from 0 to t. This is close to what you wrote, but it is like flipped on the y-axis and normalized.

- Above, I referred to the blue curve as the "Concentration". This is because the default peak settings follow the officially reported values of the peak concentration time. "Activity" of the insulin is smth else (ref. my original post). Of course, the above definition of IOB is correct: it computes the remaining insulin to be absorbed. What I was thinking is that maybe a more helpful definition would follow the peak "Activity" time, which is quite larger than the peak "Concentration" time. I would call this "Effective IOB", and would represent the "remaining insulin effect". A way to estimate the "peak activity time" is to assume it is the steepest point of the concentration -> at this point, the insulin is more rapidly used by the body. From my calculations, this produces values near those I have seen reported in research papers.

Food for thought.

MilosKozak, 2025-10-23:
@scottleibrand

scottleibrand, 2025-10-24:
The blue curve is activity: BG-decreasing effect. We have empirically tested that using autotune (with enough data, it can estimate insulinPeakTime) and find it to be close to the default values in oref0. Certainly closer to 1 hour than 2. There is a long tail, though.

Philoul, 2025-10-24:
@kbountro, the word "Concentration" should be avoided here, as it has a completely different meaning!
Insulin concentration corresponds to the number of International Units (IU) present in 1 ml of liquid.
- The most common insulin concentration is U100 (and most pumps are designed to use only U100).
- A few pens with U200 are available (e.g., Lyumjev is available in U100 and U200).
- In rare cases of extreme insulin resistance, U300 and U500 insulins can be found in hospitals in vials.
- Historically, more diluted U40 insulins have also been used in vials, with specific syringes.