nightscout/AndroidAPS #3987: Feature request : delete notifications

robertrub, 2025-05-18:
When I change my pump, I get notifications from Android and directly in AAPS.

It would be much more user friendly if, when I cleared or OKed the notification or in AAPS, the equivalent message/notification gets deleted too.

[image]

[image]

Actually, I need to OK in AAPS and then swap to delete in Android.

MilosKozak, 2025-06-05:
Clearing notification in android doesn't remove internal?

robertrub, 2025-06-05:
Tested with a profile change and yes, clearing Android notification cleared it in AAPS too. Will test on next pod change and close this issue if it works there too.

MilosKozak, 2026-10-11:
In the current dev version this works both ways. Dismissing a notification in AAPS also removes it from the Android notification bar, and the other way round. Please reopen if you still see this in 4.0.

closed 2026-10-11 by MilosKozak