nightscout/AndroidAPS #3988: no prediction lines on watch with aaps-client

maskac, 2025-05-18:
no prediction lines in CGM graph on watchface with aaps-client

Tested on version AAPS master/client verson 3.3.2.0 with Suunto 7 and Samsung Watch.

On both (master and client) app in the mobile phone prediction lines are present.

Philoul, 2025-05-21:
User should check within Watch, AAPS Settings, Graphical parameter if predictions is enabled or not...
[image]

[image]

[image]

If predictions are available within AAPSClient app on phone, they should be available within watch (but never tested myself)

maskac, 2025-05-21:
> User should check within Watch, AAPS Settings, Graphical parameter if predictions is enabled or not... !
> If predictions are available within AAPSClient app on phone, they should be available within watch (but never tested myself)

As i wrote in incial post, in aapsclient app is prediction visibile. 

I have got reported problem from parent of T1D child and replicate  it self the problem on other 2 watches/phones.

Setting on watch checked double.

olorinmaia, 2025-05-21:
I checked on my watch connected to AAPSClient app and show predictions is turned on. There is no predictions on cwf.

Philoul, 2025-05-21:
Ok, so it's an issue 😉
I will try to see the trigger/information used to send predictions to the watch to understand why they are not included.

Maybe @olorinmaia you can help me if you have a test conf where you can reproduce the issue.

olorinmaia, 2025-05-23:
Fixed in dev with commit 7011790
@MilosKozak this issue can be closed.