nightscout/AndroidAPS #4032: Feature request : Color lines in treatments

robertrub, 2025-06-15:
It would help the users know what these colors are
[image]
if the equivalent line in the treatments running mode tab had the same color.
[image]

Maybe (also ?), a tap on the colored bar on the UI can have an explanation pop-up to tell what the color means.

If not, we'll have to answer the same questions over and over in Discord... 🫣

robertrub, 2025-06-15:
Yes, sounds logical. I don't know what the actual colors mean. Only had red (time before the update) and yellow (pump change).

I vote against the **green line** though. The UI would be cleaner if there were no lines if loop is working normally (closed loop, which is 99% of the time I suppose) and show a color only when it is not.

olorinmaia, 2025-06-16:
Green line for closed loop was left out intentionally to avoid cluttering up graph when loop is in normal state. But I think it's a nice improvement to add text when you tap the color-lines :)