nightscout/AndroidAPS #4036: How to Export All Data from AAPS (CGM, Basal, Bolus, IOB, etc.)

leedabid, 2025-06-17:
Dear AAPS team,

I would like to export all data from AAPS, including:
 - CGM values (timestamps and readings)
 - Basal insulin rates and their timestamps
 - Bolus insulin doses and types (SMB vs. user-entered), with timestamps
 - IOB data
 - And any other logs related to therapy decisions

I tried using the following menu path:
Hamburger Menu (≡) > Maintenance > Miscellaneous > EXPORT USER ENTRIES TO EXCEL (CSV)

This generated a .csv file at:
AAPS > temp > DB > exports

However, this file seems to include only user-entered log data (such as notes or manual entries), not the full dataset I am looking for.

Could you please advise:

Is there a way to export the complete dataset, including CGM, basal, bolus (SMB/user), and IOB data?

If not through the UI, is there a recommended method (e.g., database access, Nightscout sync, or API) to extract this data?

Philoul, 2025-06-17:
Nightscout is the "backup" database.
If you switch loop on a new phone, a full sync will restore most of your data. (BG values, carbs, boluses, TBRs, Careportal events, profile switches...)
IOB/COB are recalculate dynamically

leedabid, 2025-06-27:
Thank you for your response.

I have a follow-up question.
To export all data related to a user's blood glucose, insulin, and meal entries (in formats such as .csv, .xlsx, or .json), is it absolutely necessary to use Nightscout?

For users who do not use Nightscout, is there any way to directly download this data from the AAPS app on their phone?

Also, I would like to know if there is a way to export logs of user actions or interactions with AAPS—such as settings changes, manual overrides, or bolus confirmations.

Philoul, 2025-06-27:
@leedabid within AAPS, you can export all user entries in csv (Compatible with Excel)
You can launch thîs exports from Maintenance or from Treatments > UE icon on the right > 3 dots menu

ga-zelle, 2025-07-11:
@leedabid My situation is similar to yours in that I don't use Nightscout. I extract all the information I want from the AAPS logfiles. But creating and maintaining scripts for that is tedious, e.g. quasi permanent updating with every version/formatting change.

andyrozman, 2025-07-12:
Are you trying to export all the data to take a look at it, or do you just wish to transfer the data to different AAPS instance?