nightscout/AndroidAPS #4052: Accu-Chek Combo Unable to pair

qq254344379, 2025-07-08:
Hello, my mobile phone is Android15，Currently using AAPS 3.3.0DEV.After I change this phone, I can't connect with it.Accu-Chek Combo Unable to pair，I have tried to disable the full-screen notification display myself，Still no effect.

https://github.com/user-attachments/assets/f705d95e-3547-45b0-a834-8582b1ff012a

[AndroidAPS_LOG_1751988502378.log.zip](https://github.com/user-attachments/files/21124976/AndroidAPS_LOG_1751988502378.log.zip)

qq254344379, 2025-07-09:
> 您是否使用“my BMW”应用程序？我们已经从那个应用程序看到其他泵（Medtrum 和 DASH）受到奇怪的干扰..

I'm sorry, I don't have this software. It should be that the window was full-screen when the scan was visible in advance, which led to AAPS misunderstanding and quit.

qq254344379, 2025-07-09:
[image]

qq254344379, 2025-07-09:
@MilosKozak

closed 2025-07-14 by qq254344379

salgol, 2026-09-10:
Had the same issue, worked out by rotating the phone horizontally so the popup widow needs to be scrolled is not closed.