nightscout/AndroidAPS #4064: Question about delta calculations - is it weighted average?

Kaszub09, 2025-07-20:
Hi, I have a question about 15 min delta calculations (and the same about 40 min).
Initially, according to [docs](https://androidaps.readthedocs.io/en/latest/DailyLifeWithAaps/AapsScreens.html#current-blood-glucose)  I understood is as:
15Δ = average of three Δ in last 15 minutes.
In other words, difference between current reading and 15 min ago divided by 3.

But from what I see, it's average of averages? Consider following (real) example from logs
```
12:26:15.983 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753006823113, value=144.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=5 avgDelta=-4.0
12:26:15.983 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753006523113, value=144.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=10 avgDelta=-2.0
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753006223113, value=145.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=15 avgDelta=-1.6666666666666665
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753005923113, value=142.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=20 avgDelta=-0.5
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753005623113, value=144.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=25 avgDelta=-0.8
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753005323113, value=144.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=30 avgDelta=-0.6666666666666666
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753005023113, value=143.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=35 avgDelta=-0.4285714285714286
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753004723113, value=142.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=40 avgDelta=-0.25
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():79]: InMemoryGlucoseValue(timestamp=1753004423113, value=139.0, trendArrow=NONE, smoothed=null, filledGap=false, sourceSensor=LIBRE_2) minutesAgo=45 avgDelta=0.11111111111111112
12:26:15.984 [main] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():264]: Glucose: 140 mg/dl Noise: 0 Delta: -4 mg/dlShort avg. delta:  -2,56 mg/dl Long avg. delta: -0,53 mg/dl Dura ISF minutes: 45,00 m Dura ISF average: 142,70 mg/dl Parabola minutes: 15,00 m Parabola correlation: 0,92 Parabola fit a0: 140,25 mg/dl Parabola fit a1: -3,75 mg/dl/5m Parabola fit a2: -0,75 mg/dl/(5m)^2
```
We have BG:
BG0 (0min)=140
BG1 (5min)=144
BG2 (10min)=144
BG3 (15min)=145

According to logs, we have avgDelta 5min=-4, avgDelta 10min=-2, avgDelta 15min=-1.66
But then 15Δ displayed on screen is =-2.56 (from logs: Short avg. delta:  -2,56)? Which is average of those three avgDelta? 
https://github.com/nightscout/AndroidAPS/blob/master/implementation/src/main/kotlin/app/aaps/implementation/iob/GlucoseStatusProviderImpl.kt 

So it basically functions as weighted average, where delta from BG1-BG0 has weight 3, BG2-BG1 weight 2, and BG4-BG3 weight 1?
 
I'm just curious if this is intended?

sanderrenes, 2025-07-29:
The short answer, it is taking the average over calculated deltas. This is an open issue:

https://github.com/nightscout/AndroidAPS/issues/3933

scottleibrand, 2025-09-24:
If you wanted to use the 1.66 value, you would be throwing away the information from the intermediate readings. By doing the average of average deltas, those readings do actually influence the result. 

The method we used for coming up with these algorithms was extremely iterative and empirical: we would try something simple, see if it worked well, and if necessary add additional data to cover corner cases. I don't remember the specifics of this one from 10 years ago, but most likely we averaged the multiple readings to get smoother behavior while avoiding under-reacting to BG drops or over-reacting to short spikes upward. 

For your purpose of extending the algorithm to every-minute CGM readings, you'll probably need to consider how jumpy the every-minute data is, and whether you'll get better results from using the more granular data, or if it's sufficient to use it to create sliding 5m windows of data that update slightly every minute (and then apply the current algorithms to the 5m-bucketed data).