nightscout/AndroidAPS #4097: Bolus calc : if pump gives error, the carbs are not registered

robertrub, 2025-08-13:
This bug happened some time ago and happened again 2 days ago. Pump Nano. Recent dev.

Used calculator, 40g carbs and 7U insulin. Insulin starts being given. As I was sick and had used a lot of insulin, the pump or AAPS trigered an alert saying that daily dose of insulin limit was given and either disconnected pump or put in open loop, can't remember exactly.

The bug : some part of the insulin was given and correctly registered but the carbs were not registered. That is not a correct behaviour.

Feature request : It would be better if AAPS checked the insulin security limits when the user validates the calc page, before starting to inject insulin (compared to actual state : inject but kick an alert in mid injection)

Sorry, no logs.

MilosKozak, 2025-08-14:
this is by design to prevent double carbs entry on 2nd try

MilosKozak, 2025-08-14:
There IS a messaghe you shoud check insulin and carbs