nightscout/AndroidAPS #4133: Feature request: SMB enable/disable button

jacekraf, 2025-09-04:
What if we could have a special button "Enable/disable SMB" button on the main screen, just like Insulin, Carbs, Calculator, CGM?

I see many people disabling SMB at night so this might be a convenient feature for them.

Philoul, 2025-09-06:
I often see these kind of requests in discord too, but not convinced...

Probably a topic to be discussed here 🤔.

In my mind, SMBs should never be considered as the root cause of hypo, especially  during nights. Your first thoughts should allways be to update/fix your profile instead of disabling SMBs.

t1dude, 2025-09-08:
If you want to temporarily disable SMB's, you could just set a high temp target and adjust your settings accordingly so that SMB's are disabled with a high TT. High does not have to mean **high**, just higher than your normal target.

MilosKozak, 2025-09-17:
https://github.com/nightscout/AndroidAPS/commit/b96e957addb7654622e4a147c62a314b62773fd3

closed 2025-09-17 by MilosKozak