nightscout/AndroidAPS #4134: AAPS CI keeps failing (dev branch)

vanelsberg, 2025-09-08:
08/09/2025:
For the past week AAPS CI build on the dev branch keeps failing.

[image]

[image]

qq254344379, 2025-09-08:
I also encountered a situation where the browser could not compile Dev.

[image]

vanelsberg, 2025-09-09:
> I will adjust it so that AAPS-CI builds the app first, then the wear, which should prevent this problem.

Interesting. Wil have a look.
Actually this should not be a problem. My Docker Autobuild solution (uses command line build) can still build App en Wear-app in a single build without problems.