nightscout/AndroidAPS #4168: QuickWizard correction

infossk, 2025-09-19:
Hi, 
please check: In QuickWizard, we have a different correction percentage set than in the global setting. However, when feeding through a "pretzel", it takes the global setting into the calculation.

Thx.

Marián

[image]
[image]
[image]

olorinmaia, 2025-09-19:
Hi. I tried to reproduce this, but I'm not able to on latest dev or 3.3.2.1

infossk, 2025-09-20:
I recorded in AAPS Client, but it's the same on mobile in AAPS.

https://github.com/user-attachments/assets/7f7fd7c5-eeb9-43a4-81a4-2a517440d415