nightscout/AndroidAPS #4182: AAPS Dev/Debug - LeakCanary results

vanelsberg, 2025-09-27:
Below info on **LeakCanary** (LC) results:

- AAPS Logs
- LC shared info .txt
- Other

vanelsberg, 2025-09-27:
Interesting: Just now LeakCanary got active whenever I start changing Automations?

[20250927-1655-3.zip](https://github.com/user-attachments/files/22575505/20250927-1655-3.zip)

[image]

vanelsberg, 2025-09-28:
Latest dev/debug build#5375e401 generated these Leak results during the past 6 hours.
First one in list is continuesly updating every few minutes:

[shared_text.txt](https://github.com/user-attachments/files/22580913/shared_text.txt)

[AndroidAPS_LOG_1759045522082.log.zip](https://github.com/user-attachments/files/22580915/AndroidAPS_LOG_1759045522082.log.zip)

[image]

vanelsberg, 2025-09-29:
AAPS dev, commit #5375e401, Sep 27,2025:

@MilosKozak:  LeakCanary txt tracefiles for latest dev commit:

[1.zip](https://github.com/user-attachments/files/22604170/1.zip)
[2.zip](https://github.com/user-attachments/files/22604172/2.zip)

*Logfiles*
[AndroidAPS_LOG_1759163252556.log.zip](https://github.com/user-attachments/files/22604163/AndroidAPS_LOG_1759163252556.log.zip)

MilosKozak, 2025-09-29:
Equil Pairing:
```
 D  ====================================
 D  HEAP ANALYSIS RESULT
 D  ====================================
 D  0 APPLICATION LEAKS
 D  
 D  References underlined with "~~~" are likely causes.
 D  Learn more at https://squ.re/leaks.
 D  ====================================
 D  0 LIBRARY LEAKS
 D  
 D  A Library Leak is a leak caused by a known bug in 3rd party code that you do not have control over.
 D  See https://square.github.io/leakcanary/fundamentals-how-leakcanary-works/#4-categorizing-leaks
 D  ====================================
 D  5 UNREACHABLE OBJECTS
 D  
 D  An unreachable object is still in memory but LeakCanary could not find a strong reference path
 D  from GC roots.
 D  
 D  android.widget.LinearLayout instance
 D  ​  Leaking: YES (ObjectWatcher was watching this because app.aaps.pump.equil.ui.pair.EquilPairAssembleFragment received
 D  ​  Fragment#onDestroyView() callback (references to its views should be cleared to prevent leaks) and View.mContext
 D  ​  references a destroyed activity)
 D  ​  key = f47ab0d0-c324-48bc-be3c-93b74d46fddf
 D  ​  watchDurationMillis = 101089
 D  ​  retainedDurationMillis = 96089
 D  ​  View not part of a window view hierarchy
 D  ​  View.mAttachInfo is null (view detached)
 D  ​  View.mWindowAttachCount = 1
 D  ​  mContext instance of app.aaps.pump.equil.ui.pair.EquilPairActivity with mDestroyed = true
 D  
 D  android.widget.LinearLayout instance
 D  ​  Leaking: YES (ObjectWatcher was watching this because app.aaps.pump.equil.ui.pair.EquilPairSerialNumberFragment
 D  ​  received Fragment#onDestroyView() callback (references to its views should be cleared to prevent leaks) and View.
 D  ​  mContext references a destroyed activity)
 D  ​  key = 8be8dc46-406d-4ca6-97fc-46613ba2be4f
 D  ​  watchDurationMillis = 69737
 D  ​  retainedDurationMillis = 64736
 D  ​  View not part of a window view hierarchy
 D  ​  View.mAttachInfo is null (view detached)
 D  ​  View.mWindowAttachCount = 1
 D  ​  mContext instance of app.aaps.pump.equil.ui.pair.EquilPairActivity with mDestroyed = true
 D  
 D  android.widget.LinearLayout instance
 D  ​  Leaking: YES (ObjectWatcher was watching this because app.aaps.pump.equil.ui.pair.EquilPairFillFragment received
 D  ​  Fragment#onDestroyView() callback (references to its views should be cleared to prevent leaks) and View.mContext
 D  ​  references a destroyed activity)
 D  ​  key = 2f16335e-910c-423b-942c-3ff1b243dd49
 D  ​  watchDurationMillis = 22900
 D  ​  retainedDurationMillis = 17900
 D  ​  View not part of a window view hierarchy
 D  ​  View.mAttachInfo is null (view detached)
 D  ​  View.mWindowAttachCount = 1
 D  ​  mContext instance of app.aaps.pump.equil.ui.pair.EquilPairActivity with mDestroyed = true
 D  
 D  android.widget.LinearLayout instance
 D  ​  Leaking: YES (ObjectWatcher was watching this because app.aaps.pump.equil.ui.pair.EquilPairAttachFragment received
 D  ​  Fragment#onDestroyView() callback (references to its views should be cleared to prevent leaks) and View.mContext
 D  ​  references a destroyed activity)
 D  ​  key = 52f53a00-0ab0-4cba-8f9e-9f6b0e6de325
 D  ​  watchDurationMillis = 21001
 D  ​  retainedDurationMillis = 16001
 D  ​  View not part of a window view hierarchy
 D  ​  View.mAttachInfo is null (view detached)
 D  ​  View.mWindowAttachCount = 1
 D  ​  mContext instance of app.aaps.pump.equil.ui.pair.EquilPairActivity with mDestroyed = true
 D  
 D  android.widget.LinearLayout instance
 D  ​  Leaking: YES (ObjectWatcher was watching this because app.aaps.pump.equil.ui.pair.EquilPairAirFragment received
[... 21 lines cut ...]
 D  Object array count: 74198
 D  Thread count: 198
 D  Heap total bytes: 73661980
 D  Bitmap count: 36
 D  Bitmap total bytes: 32588179
 D  Large bitmap count: 0
 D  Large bitmap total bytes: 0
 D  Db 1: open /data/user/0/info.nightscout.androidaps/databases/com.google.android.datatransport.events
 D  Db 2: open /data/user/0/info.nightscout.androidaps/no_backup/androidx.work.workdb
 D  Db 3: open /data/user/0/info.nightscout.androidaps/databases/androidaps.db
 D  Db 4: closed /data/user/0/info.nightscout.androidaps/databases/google_app_measurement_local.db
 D  Db 5: open /data/user/0/info.nightscout.androidaps/databases/equil_database.db
 D  Db 6: closed /data/user/0/info.nightscout.androidaps/databases/google_app_measurement_local.db
 D  Db 7: closed /data/user/0/info.nightscout.androidaps/databases/google_app_measurement_local.db
 D  Db 8: closed /data/user/0/info.nightscout.androidaps/databases/google_app_measurement_local.db
 D  Db 9: closed /data/user/0/info.nightscout.androidaps/databases/google_app_measurement_local.db
 D  Stats: LruCache[maxSize=3000,hits=265280,misses=499339,hitRate=34%]
 D  RandomAccess[bytes=25378451,reads=499339,travel=865697050237,range=73951909,size=101961582]
 D  Heap dump reason: 5 retained objects, app is visible
 D  Analysis duration: 70847 ms
 D  Heap dump file path: /storage/emulated/0/Download/leakcanary-info.nightscout.androidaps/2025-09-29_21-19-08_323.hprof
 D  Heap dump timestamp: 1759173631256
 D  Heap dump duration: 4418 ms
 D  ====================================
```

vanelsberg, 2025-09-29:
> Equil Pairing:

Equil?  - I see. To be honest, I did not check the trace txt's. 

Can you explain? (As in: how does Equil get into this as Equil was not active?)

vanelsberg, 2025-09-30:
@MilosKozak Btw: you need more of this?
If it helps, I can try running dev/debug with LeakCanary on major commits (or feel free to DM me)?

MilosKozak, 2025-09-30:
reports are welcome from people running dev. but not necessary to hunt for them specifically

ga-zelle, 2025-10-03:
Today I get leak messages. I run a virtual Pixel in AS  and the only thing I find are files of type **hprof**. On download from device manager they are somewhat readable but include binary sections, too.

@MilosKozak are those the filöes you want? Or is there an alternative, e.g. with logcat or similar?

vanelsberg, 2025-10-25:
@MilosKozak Still interested in Leak-results for latest dev commits??
If so I'll post them here...